firmata/processing #7: capitalization mismatch on library folder and file names

RangerPhoton, 2016-05-13:
note the capitalization mismatch between the folder and file names: "arduino/library/Arduino.jar"
Works fine on Processing for Mac, but Processing for Rasberry Pi didn't recognize the library until I renamed it to "arduino.jar"

soundanalogous, 2016-05-22:
Please try processing-arduino.zip in the update branch: https://github.com/soundanalogous/processing/tree/update. And let me know if that fixes the issue.

soundanalogous, 2016-05-22:
I've attached the updated zip for simplicity. Unzip and replace the existing /arduino/ directory in /Processing/libraries/
[processing-arduino.zip](https://github.com/firmata/processing/files/276562/processing-arduino.zip)

closed 2016-07-24 by soundanalogous