firmata/processing #8: java.lang.NullPointerException when trying to import cc.arduino.*

riis619, 2016-05-22:
Running any of the examples such as arduino_output, arduino_input, or arduino_servo I get a java.lang.NullPointerException. Also when I run a working sketch then add the line "import cc.arduino.*" I get the same error. 

The full error is:
java.lang.NullPointerException
	at processing.app.Library.getClassPath(Library.java:337)
	at processing.mode.java.JavaBuild.preprocess(JavaBuild.java:454)
	at processing.mode.java.JavaBuild.preprocess(JavaBuild.java:193)
	at processing.mode.java.JavaBuild.build(JavaBuild.java:152)
	at processing.mode.java.JavaBuild.build(JavaBuild.java:131)
	at processing.mode.java.JavaMode.handleLaunch(JavaMode.java:126)
	at processing.mode.java.JavaEditor$34.run(JavaEditor.java:1099)
	at java.lang.Thread.run(Thread.java:745)

soundanalogous, 2016-05-22:
Please try processing-arduino.zip in the update branch: https://github.com/soundanalogous/processing/tree/update. And let me know if that fixes the issue.

soundanalogous, 2016-05-22:
I've attached the updated zip for simplicity. Unzip and replace the existing /arduino/ directory in /Processing/libraries/
[processing-arduino.zip](https://github.com/firmata/processing/files/276562/processing-arduino.zip)

riis619, 2016-05-22:
Completely an issue on my end with the application being installed in a
directory then moved which caused problems. Thank you for your quick
response though.

closed 2016-05-22 by soundanalogous