firmata/processing #11: implementing MODE_INPUT_PULLUP?

jaydenrre, 2016-11-03:
https://github.com/firmata/arduino/pull/235

Does this mean that the pullup resistors can be enabled with MODE_INPUT_PULLUP?
Processing tells me that MODE_INPUT_PULLUP does not exist.

Do you have an example on how to enable / use pullup resistors with firmata and processing?

thanks

soundanalogous, 2016-11-04:
That is because [Arduino.java](https://github.com/firmata/processing/blob/master/src/Arduino.java) doesn't define MODE_INPUT_PULLUP. I can update this. In the mean time you can simply pass 11 to pinMode:

```java
arduino = new Arduino(this, Arduino.list()[0], 57600);
// Set pin D2 to INPUT_PULLUP
arduino.pinMode(2, 11);
```

jaydenrre, 2016-11-04:
Great! It works... mostly...i think...

I may have spotted a couple of bugs.

With the input example slightly tweaked it seems that there are a few pins that dont take to the MODE_INPUT_PULLUP.

In the for loop assigning the pins i added a delay and that turned on pins 9, 12, 13. But 0,1,10,11 seem unresponsive, and 8 drifts in and out. I checked with two arduinos with standard Firmata and it happens to both.

I am by no means an expert, so i could be messing something up.

thanks

`
// tweaked Firmata input example

import processing.serial.*;

import cc.arduino.*;

Arduino arduino;

color off = color(4, 79, 111);
color on = color(84, 145, 158);

void setup() {
  size(470, 280);

  // Prints out the available serial ports.
  println(Arduino.list());
  
  
  // Modify this line, by changing the "0" to the index of the serial
  // port corresponding to your Arduino board (as it appears in the list
  // printed by the line above).
  arduino = new Arduino(this, Arduino.list()[0], 57600);
  
  // Alternatively, use the name of the serial port corresponding to your
  // Arduino (in double-quotes), as in the following line.
  //arduino = new Arduino(this, "/dev/tty.usbmodem621", 57600);
  
  // Set the Arduino digital pins as inputs.
  for (int i = 0; i <= 13; i++){
    arduino.pinMode(i, 11); // 11 adds the pull up resistor mode
    delay(100); // a slight delay enables more pins?
  }
  //arduino.pinMode(0, 11);
  //arduino.pinMode(1, 11);
  
  //arduino.pinMode(10, 11);
  //arduino.pinMode(11, 11);

// when 0,1,10,11 are set to MODE_INPUT_PULLUP, they seem unresponsive

}

void draw() {
  background(off);
  stroke(on);
  
  
  // Draw a filled box for each digital pin that's HIGH (5 volts).
  for (int i = 0; i <= 13; i++) {
    
    if (arduino.digitalRead(i) == Arduino.HIGH)
      fill(on);
    else
      fill(off);
      
    rect(420 - i * 30, 30, 20, 20);
    fill(on);
    text(str(i),420 - i * 30, 70);
    noFill();
    
    
  }

  // Draw a circle whose size corresponds to the value of an analog input.
  noFill();
  for (int i = 0; i <= 5; i++) {
    
    ellipse(280 + i * 30, 240, arduino.analogRead(i) / 16, arduino.analogRead(i) / 16);
    
    fill(on);
    text(str(i),280 + i * 30, 180);
    noFill();
  }
}
`

soundanalogous, 2016-11-04:
Which board are you using?

jaydenrre, 2016-11-04:
Regular old Uno

soundanalogous, 2016-11-04:
You can't mess with pins 0 and 1 on the Uno since those are the RX and TX pins that is the communication line between your board and processing. You also can't set input pullup on pin 13 because it's wired to the on board LED (well technically you can set it but it will have no effect). Try excluding pins 0, 1 and 13 and see if that makes any difference.

soundanalogous, 2016-11-06:
I've added the INPUT_PULLUP constant to the library. You'll have to download and install the processing-arduino.zip (unzip and copy arduino file to processing/libraries/ directory) in order to get the change. I've also asked the Processing library manager to update the link to the Arduino processing github so that updates will start showing up in the Processing library manager.

closed 2016-11-06 by soundanalogous