firmata/processing #15: i2c implementation for processing?

wydmynd, 2017-06-28:
I want to build a design course around processing with physical interaction parts done using arduino and processing/firmata. 
It works great for digital/analog sensors and motors, but today many sensors and even motor drivers are i2c.
Will there be a i2c implementation for processing in the near future? thanks!

soundanalogous, 2017-06-28:
I'm not planning on adding support any time soon, but if you or anyone else is interested in contributing an implementationI'm happy to review your work. To get started, I suggest looking at how it's implemented in other firmata client libraries.

wydmynd, 2017-06-29:
Thanks for the quick reply. Unfortunately I'm not sure I'm experienced enough  to do this yet. Currently I work with Snap4Arduino+Firmata and I managed to create some new "blocks" (gui driven functions) using SYSEX functions (by modifying existing examples) . If you could refer me to an example of calling SYSEX commands from Processing that would be a great start for me. if not only to enable using things like Ultrasonic sensors. thanks again.