firmata/processing #18: release txt error

vincentsijben, 2025-08-21:
Don't know if anyone still reads this, but https://github.com/firmata/processing/releases/download/latest/processing-arduino.txt returns an error, which causes this firmata library to be broken inside Processings Manage Libraries tool...

pgrawehr, 2025-08-22:
Honestly, I don't even know what this repo is about... The link on the main page is dead, and your link doesn't work for me, either. Can you maybe explain what you're trying to do and how?

SableRaf, 2025-08-25:
TL;DR: [Processing](https://processing.org/) is a programming environment widely used in art, design, and education. The Firmata library lets Processing talk to Arduino boards. Right now the library is marked as broken because the processing-arduino.txt file is missing/invalid in the release. Adding a valid file will fix it.

Thanks @pgrawehr,

The Processing Firmata library is still often used (especially in education) to connect the [[Processing IDE](https://processing.org/)](https://processing.org/) with Arduino boards. Many [[tutorials](https://www.youtube.com/watch?v=l4jFQIoY9ZY)](https://www.youtube.com/watch?v=l4jFQIoY9ZY) and [[courses](https://www.making.unsw.edu.au/making-centre/learnresources/hackspace-learning-resources/arduinoprocessing-firmata-setup-guide/)](https://www.making.unsw.edu.au/making-centre/learnresources/hackspace-learning-resources/arduinoprocessing-firmata-setup-guide/) rely on it so people can, for example, write Processing sketches that read sensor data or control actuators.

The issue right now is that the text file Processing checks to verify a library (`processing-arduino.txt`) is missing/broken in the release. Because of this, the library is marked as [[BROKEN](https://github.com/processing/processing-contributions/blob/main/contributions.yaml#L379)](https://github.com/processing/processing-contributions/blob/main/contributions.yaml#L379) in our system, and cannot be installed or updated through Processing’s Contribution Manager—even though the `processing-arduino.zip` itself is valid.

To fix this, a valid `processing-arduino.txt` just needs to be added to the release. That will allow Processing to recognize the library again. We’d be happy to help prepare the file, or alternatively update the library to use our [[current library template](https://github.com/mingness/processing-library-template)](https://github.com/mingness/processing-library-template) and open a pull request.

zfields, 2025-08-25:
We would absolutely appreciate a PR!

That would give us something to wrap our minds around, and get processing up and running once more.

pgrawehr, 2025-08-25:
@vincentsijben It pops up on my warchlist, since I'm a firmata group member. I don't know all of the repositories in the group,  though.

vincentsijben, 2025-08-29:
update..

[image]

vincentsijben, 2025-09-29:
Great news:

> Our team has let me know that they deployed a fix for this and the file you shared can now be retrieved.

It works now 🥳

pgrawehr, 2025-09-29:
@vincentsijben So is there something we need to do here now?

vincentsijben, 2025-09-29:
> [@vincentsijben](https://github.com/vincentsijben) So is there something we need to do here now?

nope. somehow, they fixed it. the link works now. 
it already was picked up by the Processing contribution https://contributions.processing.org/contribs.txt

[image]